KitronikLtd/Kitronik-Pico-Robotics-Board-MicroPython #2: Continuous rotation servo support

shepai, 2022-07-12:
Hi, I was reading through the datasheet and it suggested that the board supports continuous rotation servos. I was wondering how to do this? I currently am writing to the servos and only being able to move between 0 and 180. Is there a way to have a speed between -1 and 1?

Thanks in advance :)

AlasdairAtKitronik, 2022-07-13:
Hi @shepai, controlling a continuous rotation servo still uses '0-180' as in an put, with 90 being stationary, 0 full speed one direction and 180 being full speed the other direction. This should work with the module provided. Have you tried this with continuous rotation servos and it just hasn't worked at all?

shepai, 2022-07-13:
Hi @AlasdairAtKitronik 
Thanks for your quick reply! Ah yes, works fine. It seems the motors I had were attached to a chassis that required more torque than they could give out. When I detached them and ran the code, the servos moved continuously. Thank you :)

closed 2022-07-13 by shepai

AlasdairAtKitronik, 2022-07-13:
@shepai Glad it's all sorted :)