KitronikLtd/Kitronik-Pico-Robotics-Board-MicroPython #3: ServoWrite: It doesnt stay in the degree where I wanted, instead it goes there and comes back

Hsgngr, 2022-07-24:
```
for degrees in range(0,90,1):
                board.servoWrite(3,degrees)
                utime.sleep_ms(5)
```

This code rotates servo to 90 degrees and it comes back again to its original position again. How can I make them stay on the degree that I wanted them to stay.

AlasdairAtKitronik, 2022-07-26:
Hi @Hsgngr, I'm not able to replicate your issue when running your code in this example program:
```python
from PicoRobotics import KitronikPicoRobotics
import utime

board = KitronikPicoRobotics()

for degrees in range(0,180,1):
    board.servoWrite(3,degrees)
    utime.sleep_ms(5)
```

Please could you send the full program your code is part of so we can see the context? Thanks.

Hsgngr, 2022-07-26:
Yes sure here is the code:

```
while True:
    led = Pin(25, Pin.OUT)
    led.toggle()
    time.sleep_ms(20)
    if uart.any() > 0:
        rxData = uart.read()
#         time.sleep_ms(20)
        print(rxData)
        data = (rxData.decode("utf-8")).replace("\n","") #Convert to readable format (byte to string)
        
        if "right" == data:
            uart.write(rxData)           
            for degrees in range(30,60,1):
                board.servoWrite(1,degrees)
                utime.sleep_ms(10)
            
        elif "left" == data:
            uart.write(rxData)
            for degrees in range(60,30,-1):
                board.servoWrite(2,degrees)
                utime.sleep_ms(10)
```
I did what I wanted to do in the end. Instead of changing the degrees if I give enough time  for example `sleep_ms()` and give the same degree regardless of `degrees` in for loop it goes to that degree.

Example:
```
            for degrees in range(60,30,-1):
                board.servoWrite(2,30)
                utime.sleep_ms(10)

```
The only change I did between the previous one was giving the same degree (a hard coded one) instead of iterated `degrees`.

I see that [simplyServo](https://github.com/KitronikLtd/Kitronik-Pico-Simply-Servos-MicroPython/blob/1575517ad2d34878c997701630ab4f73ee5203eb/simplyServos.py#L44) board has this feature `goToPosition()` but Robotics board doesnt have yet. The issue that I have

AlasdairAtKitronik, 2022-07-27:
@Hsgngr Thanks for sending your code through.

I think you've got a couple of issues going on here:

Firstly, because your code runs forever in the 'while True' loop, if the value of 'data' remains either "right" or "left", the servo will continually be sent the command to move from 30 degrees to 60 degrees (or the opposite for servo 2). This means it will constantly be moving 30 degrees one way and then the other as it repeats the instruction. In order to make sure the instruction doesn't keep being sent to the servo, a flag needs to be set the first time it is carried out. When the opposite command is sent over UART, the flags need to be switched over. The example code below shows this:

```python
from PicoRobotics import KitronikPicoRobotics
import utime

board = KitronikPicoRobotics()
rightFlag = False # Both flags are False to start with as no "left" or "right" commands have been sent yet
leftFlag = False

while True:
    
    if ("right" == data) and not(rightFlag): # Check the the rightFlag is not already set to True
        rightFlag = True # Set the rightFlag to True to stop repeated runs through the loop
        leftFlag = False # Set the leftFlag to False so that servo 2 can be controlled if the "left" command is sent again
        uart.write(rxData)
        for degrees in range(30,60,1):
            board.servoWrite(1,degrees)
            utime.sleep_ms(10)
        
    elif ("left" == data) and not(leftFlag):
        leftFlag = True
        rightFlag = False
        uart.write(rxData)
        for degrees in range(60,30,-1):
            board.servoWrite(2,degrees)
            utime.sleep_ms(10)
```

Secondly, the servos appear to be turning to the angle you want and then back again, but actually, they are returning to the starting position and then carrying out the command as instructed. For servo 1, the steps are as follows:
``` python
for degrees in range(30,60,1):
            board.servoWrite(1,degrees)
            utime.sleep_ms(10)
```
- Set servo 1 to 30 degress
- Pause 10ms
- Set servo 1 to 31 degrees
- Pause 10ms
- (continue around 'for' loop)
- Set servo 1 to 60 degrees
- Pause 10ms
- Exit 'for' loop

Servo 1 will remain fixed at 60 degrees, but as soon as the "right" command is sent again, it will jump back to 30 degrees as the 'for' loop starts again. Unless you reset the servo angles somewhere else in the code, they will always move back and forth with each run through the 'for' loop. I'm not sure what the physical setup is that you're using these servos in, but one option could be to reset the servo angle when the other command is sent, as shown in the example code below:
```python
if ("right" == data) and not(rightFlag):
        rightFlag = True
        leftFlag = False
        uart.write(rxData)
        board.servoWrite(2, 60) # Set servo 2 back to its starting position of 60 degrees, ready for next "left" command
        for degrees in range(30,60,1):
            board.servoWrite(1,degrees)
            utime.sleep_ms(10)
        
    elif ("left" == data) and not(leftFlag):
        leftFlag = True
        rightFlag = False
        uart.write(rxData)
        board.servoWrite(1, 30)  # Set servo 1 back to its starting position of 30 degrees, ready for next "right" command
        for degrees in range(60,30,-1):
            board.servoWrite(2,degrees)
            utime.sleep_ms(10)
```

**Answering points in your second message:**
In the second example piece of code you sent:
```python
for degrees in range(60,30,-1):
            board.servoWrite(2,30)
            utime.sleep_ms(10)
```
The 'for' loop has no impact at all, the servo position will just be set to 30 degrees every time.

Finally, in the 'simplyServo' module, the 'goToPosition()' function has identical functionality to the 'servoWrite()' function here.

Hopefully this helps!

Hsgngr, 2022-07-28:
Thanks for the detailed comment. 

The loop is infinite however the data comes just once so there is no need for flag. 
 But when I use board.servoWrite(1,60) it works actually. I remember why I added the for loop  (for the speed regulation).

closed 2022-07-28 by Hsgngr