KitronikLtd/Kitronik-Pico-Robotics-Board-MicroPython #6: Board limited to 2 servos, but how many can the Pico (W) run in total?

sgbaird, 2022-10-08:


DaveAtKitronik, 2022-10-10:
This board is designed to operate 4 DC motors at up to 10V. It will also operate up to 8 servos (at the same time as the motors) although it is usual to run it at 5-6V for that as servos generally do not want 10V. 
You could operate up to 8 pumps using this board and 8 RC speed controllers to drive the pump motor - something like this: https://hobbyking.com/en_us/turnigy-30a-brushed-esc.html 
The motor speed ( and hence flow) is controlled via the 'servo position'.
Depending on the speed controller it can be 0% power at 0 degrees to 100% power at 180, or 0% power at 90 degrees, with +100% at 180 and -100% at 0 degrees (motor can go forwards and reverse.)
Using RC speed controllers you could also use the ['Simply Servos](https://kitronik.co.uk/5339)' board, driver software for that is here: https://github.com/KitronikLtd/Kitronik-Pico-Simply-Servos-MicroPython

Hope that helps, closing this as it seems to be complete.  
If you require further assistance then contact us [here](https://kitronik.co.uk/pages/contact-us) 
Dave

closed 2022-10-10 by DaveAtKitronik

sgbaird, 2022-10-10:
Hi Dave, thank you for clarifying! I'll keep the suggestion about the servo power controllers in mind.