KitronikLtd/Kitronik-Pico-Robotics-Board-MicroPython #7: Wifi never get activated on RP2 W in standalone power (from Vsys Kitronik board)

felinh, 2023-03-05:
Hi, I am using the RP2 is hosted on the Robotics [Board](https://kitronik.co.uk/products/5329-kitronik-compact-robotics-board-for-raspberry-pi-pico) and running [[micropython-mqtt](https://github.com/peterhinch/micropython-mqtt)] on MicroPython v1.19.1-915-g2bcd88d55 on 2023-03-02; Raspberry Pi Pico W with RP2040.
Every thing is working well when   RP2 is powered from USB while plugged in the board (that is powered too): it works up to the mqtt connection .
But when powering RP2 with Vsys only (thru kitronik VSYS board), it apprears that `self._sta_if.active(True) `is not effective at all (I printed `self._sta_if.active() `that always return [False](url)).

I have opened a github issue on mqtt side  #[1610024776](https://github.com/peterhinch/micropython-mqtt/issues/107#issue-1610024776), but it is possible that there an electrical conflict at HW level (Robotics LED board brigthness seems to decrease on WIFI attempts), and thus to diagnose, schematic of the Robotics board would be necessary. 
I have two instances of this board, it is very compact and suitable for my need, but I can't use Wifi capability due to this issue.
Can you help me please?

JackAtKitronik, 2023-03-06:
Hi there,

Thanks for your message and I am sorry to hear you are having an issue with the board. It is an issue we are aware can crop up and are working to resolve for our customers as quickly as possible. The regulator on the Pico board, whilst notably above specification for normal Pico use, runs into difficulties when trying to boost the WiFi to achieve connection as the current spike causes the regulator to kick in to protect the circuit. Unfortunately this board was created before the Pico W came out and so hadn't taken into account the current spike caused by the WiFi chip.

We are currently in the process of upgrading and changing the regulator to one that can handle the additional current when connecting. We are hoping that this can be done within the next few weeks.

In the meantime if you power the Pico separately from a USB during the connection period that can act as a work around in the short term for you.

Feel free to send us an email if you'd like more help with this issue: admin@kitronik.co.uk

Apologies again for the inconvenience caused.

felinh, 2023-03-06:
Hi 
As a temp fix, I did isolated the RP2 VSYS from the Kitronic board , added a schotty and connected to a 5V buck. This is absolutely dirty but it works.
I hope this won't cause any side effect, can you please confirm?
[image]

DaveAtKitronik, 2023-03-06:
What you have done shouldn't cause any problems. 
There is already a schotty diode on the board between the regulator and the Pico VSYS pin, so you shouldn't need to isolate the pin from the board.

slabua, 2023-03-19:
I was about to comment about this too. I just got the board few days ago and I will not get the new revision.
Just to confirm, is it safe to supply power to both the board (10.8V max) and the pico via usb at the same time, without any modification?

DaveAtKitronik, 2023-03-20:
Yes, you can power up the board with upto 10.8V at the same time as powering the Pico via its USB connector. 
The onboard regulator is diode linked into the Pico.

closed 2023-06-30 by JackAtKitronik