KitronikLtd/Kitronik-Pico-Robotics-Board-MicroPython #10: Pico W not connecting

cbrmichi, 2024-01-20:
Hi,
I´m using a pico W and stuck at connecting the board to the pico w. I just wanted to try the provided test-
files.
The pico is connected and powered by USB while the board is powered by battery (6v). The onboard LED is on.

Code:
```
import PicoRobotics
import utime

board = PicoRobotics.KitronikPicoRobotics()
```

I´m getting this error:

  File "<stdin>", line 4, in <module>
  File "/lib/PicoRobotics.py", line 189, in __init__
  File "/lib/PicoRobotics.py", line 18, in initPCA
OSError: [Errno 5] EIO

So i tried i2c.scan(), which gave me [108, 112]. So it should be connected and working, but it doesn´t. 
Could you please tell me, what is wrong?

Thanks in advance,
Michael

user-redans, 2024-02-07:
Hi, @cbrmichi 

I also have the same issue. Did you find a solution?

Thanks in advance.

cbrmichi, 2024-02-07:
> <KitronikLtd/Kitronik-Pico-Robotics-Board-MicroPython/issues/10/1932155629
> @github.com>

Hi!

At first i tried using circuitPython. That did work, but i wanted to use
microPython. The support (thanks again!) suggested reinstalling the newest
microPython (w!) firmware. I did that before having tried circuitPython,
but did it again next day and it worked immediately.
From this on the problem was solved.

Maybe changing between the Firmware helped...

And don't forget to check the power switch on the board :)

Hsgngr, 2024-04-11:
Having the same problem, which version are you currently using ?

jkozniewski, 2024-04-28:
Same here - tested with latest official build v1.22.2 and latest preview build v1.23.0-preview.343.ge60e8079a with no success, is the boar just incompatible with pico W ?

felinh, 2024-11-07:
duplicate of #7 :  Kitronik said :"Unfortunately this board was created before the Pico W came out and so hadn't taken into account the current spike caused by the WiFi chip.", workaround explained in #7

JakeAtKitronik, 2025-01-02:
Apologies for the delay in responding. 

Although it is not recommended to use a Pico W with this Robotics board, we have internally tested the Pico Robotics board with a Pico W running the latest MicroPython (v1.24.1). 

Unfortunately, we were not able to reproduce the error.

It appears that there was an ongoing issue with I2C on the Pico W under MicroPython, which seems to have been resolved in later releases. 

Please update the version of MicroPython that's running on the Pico, and let us know if there are any further issues!
Kind regards, 
Jake

closed 2025-01-02 by KitronikLtd