KitronikLtd/Kitronik-Pico-Robotics-Board-MicroPython #15: Is it possible to change the I2C pins the Pico Robotics Board communicates with

RnoldR, 2025-01-17:
The Pico has two I2C controllers: 0 and 1. According to the documentation "The Pico pins GP8 and GP9 are used for the I2C bus". That means controller 0. Any other device using the pins of controller 0 (eg. pins GP2 and DP3) get the error message that the device is already in use. They should for example use pins GP 4 and GP5, thus using I2C controller 1. 

I recently tried the Pico plus with the motor board and it used GP2 and GP3 for their Stemma/QT breakout. When I connect a device to this breakout I'll get the infamous message that the device is already in use. Is it possible to use other pins so I can select controller #1? I examined the code in the PicoRobotics.py driver and found some code commented out that referred to this possibility. But of course the hardware should accomodate this.

Any suggestions?

RnoldR, 2025-01-17:
Messed with the pin numbers, corrected them.

JakeAtKitronik, 2025-01-17:
Just to confirm, are you trying to add additional I2C devices onto controller 0 by defining the pins GP4 and GP5 as I2C?

RnoldR, 2025-01-17:
1. I try to connect a vl53l1x to the stemma/qt plug of the raspberry pico plus. This plug is connected to GP4 and GP5 (see: pinout: https://shop.pimoroni.com/products/pimoroni-pico-plus-2?variant=42092668289107). 
2. Alas no, I just corrected some typo's. The problem as described now is ok.

RnoldR, 2025-01-17:
The way I now connect the vl53l1x is by connecting it to GP14 and GP15. It works but bypasses the stemma/QT breakout.

JakeAtKitronik, 2025-01-17:
Right, got it. 
Unfortunately, he design of the Robotics board does not allow for the board to be driven from controller 1 instead of controller 0. 

I would advise soldering a pin header to the breakout on the robotics board underneath the Pico, and connecting the Time of Flight sensor to GPIO 2 and GPIO 3 (I2C1) - please note that the silkscreen under the board refers to the physical pins on the Pico, rather than the GPIOs - meaning you need to connect to pin 4 (GPIO2) and 5 (GPIO3) on the robotics board.

RnoldR, 2025-01-21:
@JakeAtKitronik, I already used stacking headers for the Pico 1 and 2 so they are being used on the Plus as well. For a next generation of the Robotics board you might consider using dipswitches for selecting a different I2C port. It's a fun board to use after all.

closed 2025-01-21 by RnoldR