KitronikLtd/Kitronik-Pico-Robotics-Board-MicroPython #18: 2 steppers at the same time; not possible?

Bfreak, 2025-01-27:
Hi. As far as I can tell, using the examples provided, there's no way to control two stepper motors at once, is that right? Thanks.

JakeAtKitronik, 2025-01-28:
The Pico Robotics Board can support two stepper motors, stepper 1 and stepper 2:

[image]

Bfreak, 2025-01-28:
Sure, but I cannot see how a command can be written that would get them to move simultaneously

JakeAtKitronik, 2025-01-28:
Ah yes, you are correct - there is no command to move the steppers simultaneously.

Bfreak, 2025-01-28:
Thanks Jake. Is this a limitation of the board, or the code examples provided?

Bfreak, 2025-01-29:
I submitted a pull request to add multi-stepper simultaneous movement commands to your code

JakeAtKitronik, 2025-01-29:
Good morning, 
Thank you for the pull request, adding the ability to drive two steppers at the same time. 
We'll review the code internally, and we should be able to merge the pull by the end of the week. 

Although you have clearly explained the operation of your code within the comments of PicoRobotics.py, the README has not been updated. Would it be possible for you to add a brief description to it? 

Thank you, 
Kind Regards, 
Jake at Kitronik